FlaSpaceInst/EZ-RASSOR #308: Packages have unmet dependencies

agroyy, 2019-11-02:
the following packages have unmet dependencies when installing ros after cloning the repository with the --recursive command:
 ros-kinetic-ros-base : Depends: ros-kinetic-actionlib but it is not going to be installed
                        Depends: ros-kinetic-bond-core but it is not going to be installed
                        Depends: ros-kinetic-class-loader but it is not going to be installed
                        Depends: ros-kinetic-dynamic-reconfigure but it is not going to be installed
                        Depends: ros-kinetic-nodelet-core but it is not going to be installed
                        Depends: ros-kinetic-pluginlib but it is not going to be installed
                        Depends: ros-kinetic-ros-core but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
Something went horribly wrong.

CSharpRon, 2019-11-07:
I'm looking into this but I believe you need to utilize rosdep to install the dependencies that are missing from the project

@tgsachse is this a manual step or was this included in the installation script?

tgsachse, 2019-11-07:
The script uses rosdep. We need more information. Are you installing Ros manually (thru the ROS wiki instructions) or are you using the installation script in this package?

tgsachse, 2019-11-07:
In the readme (https://github.com/FlaSpaceInst/EZ-RASSOR/blob/master/docs/README.rst#typical-installation) for the typical installation notice that it says if you get certain dpkg errors there's a command you might want to run to fix it.

Dpkg/apt just randomly fails when installing ROS sometimes. Something to do with package conflicts etc. It's pretty junk tbh but it's out of our control :(

CSharpRon, 2019-11-11:
@agroyy is this still an issue or can we go ahead and close?

agroyy, 2019-11-11:
Should be good to close.

CSharpRon, 2019-11-11:
Closing issue since ROS dep needs to be used.

closed 2019-11-11 by CSharpRon