FlaSpaceInst/EZ-RASSOR #309: Add in Continuous Integration (CI) with Jenkins

CSharpRon, 2019-11-07:
It is about time that we add in CI to the repository. The functionality of the CI would be as follows:

1) Watch for Pull Requests to the [Staging](https://github.com/FlaSpaceInst/EZ-RASSOR/tree/staging) branch
2) Download our repository
3) Download an Ubuntu 18.10 Docker image
3) Unit Test our code inside the container using [Python unittest](https://wiki.ros.org/unittest) (note: this would require a review of our existing ROS nodes and most likely some modifications to the code to support unittest)
4) Return the result of testing as a comment to the PR (perhaps we could even decide to automatically deny the PR).
5) Display a CI status tag on our README.   

I intend on getting this work done before the year is up. Some of the implementation details may need to change depending on what's discussed here.   

---   

If you are interested in why I decided on [Jenkins](https://jenkins.io/) vs [Travis CI](https://travis-ci.com/), Jenkins is truly open source(OS) and allows for Self-Hosting. Travis CI is easier to use with GitHub and free for OS projects, but I would rather have control of where we are hosting CI. Check this [breakdown](https://stackoverflow.com/questions/32422264/jenkins-vs-travis-ci-which-one-would-you-use-for-a-open-source-project#answer-37769209) out for more information.

CSharpRon, 2019-11-16:
Actually, GitHub Actions functionality was just added to this repo. Let's do that instead to keep it in one place :) (aka, no need to host build software on a different machine)

tgsachse, 2019-12-25:
Yeah I think auto-closing PRs that don't pass the tests is fair (if that is possible).

PR #346 (2020-07-24): Staging -> Master: GitHub Actions and PEP8 changes
Another milestone complete!
Closes #309 #311
closed 2020-07-24 by CSharpRon