FlaSpaceInst/EZ-RASSOR #311: Easier Style Compliance

rmorshea, 2019-12-11:
Heard on [the changelog podcast](https://changelog.com/podcast/372) keeping PEP8 compliance was done manually during review. I'd highly recommend using Black to automatically format your code for you in a PEP8 compliant manner:

https://black.readthedocs.io/en/stable/

CSharpRon, 2019-12-24:
@rmorshea that's a great tool. Good catch on our other issue. I'm working that now and you're right, it does look like a perfect tool to introduce into our CI pipeline. I'll post my findings here before closing out the issue.

CSharpRon, 2020-07-24:
@rmorshea it's been a long while but thank you again for the recommendation.

I manually ran Black to format the repo code and this ultimately led to me using Flake8 in our CI process to automatically tell us if the code if PEP8 compliant. It will also provide annotations so we can see where the changes need to be made.

We also decided against making Black a part of the CI process, I would rather tell the developer which changes to make instead of making it for them and potentially breaking something. 

Just thought you should know. Again, thanks for sparking the conversation. It contributed to the changes we made.

closed 2020-07-24 by CSharpRon

rmorshea, 2020-07-25:
@CSharpRon it may be worth noting that `black` ensures the AST remains unchanged following a reformat. This ensures that `black` cannot accidentally "break something" because the actual logic is perfectly retained. I'd give `black` another look if you weren't aware of this before. At least in my experience, nitpicking code-style in reviews is never really fun - it's not fun as a reviewer and it's not really fun as a contributor. It's way easier to just have the computer do the review for you.

If remembering to apply `black` feels burdensome for developers, maybe check out [pre-commit](https://pre-commit.com/) which can automate this (and other things) each time you commit code. Here's a reference for [one of my personal projects](https://github.com/idom-team/idom/blob/master/.pre-commit-config.yaml).

In any case though, whether or not you end up using `black`, I'm happy this drove some positive change.

Good luck going forward.