FlaSpaceInst/EZ-RASSOR #323: Mobile App not working

gilchristb20, 2020-03-10:
The mobile app listed under EZ RASSOR Controller is not updated so it is unable to be downloaded. when you click the link to download [here](https://expo.io/@camilozano/ControllerApp), Expo states that the project is out of date and must be updated from Expo SDK v31.0.0 to Expo SDK v33.0.0

CSharpRon, 2020-03-10:
Apologies, we created a separate account in Expo and the link you have should be deprecated. Use https://expo.io/@ez-rassor/EZRASSORController instead

Also, where does that link exist in the documentation? Is it in the Wiki?

CSharpRon, 2020-03-10:
Also, the app is live on the Play Store https://play.google.com/store/apps/details?id=com.FlaSpaceInst.EZRASSORController

gilchristb20, 2020-03-10:
the link is located in the wiki under EZ RASSOR Controller

CSharpRon, 2020-03-10:
Thank you, just need a few hours before I'm back online to fix the link. Then I'll close the issue

CSharpRon, 2020-03-10:
Updated the wiki and the links are live. Thank you for the heads up @gilchristb20 !

closed 2020-03-10 by CSharpRon