FlaSpaceInst/EZ-RASSOR #347: Getting set up for a classroom

joerou, 2020-07-31:
Hello,

Awesome software!

I'm trying to get this set up to use in a classroom setting (for a high school level course). In an environment like that, not every student has access to a computer that they can get ubuntu running on. I was hoping for a better delivery system for the simulator and so I've been playing the last couple of days trying to get this working on a raspberry pi (mixed results) and also on a VPS using VNC (seems Gazebo doesn't like that so much). I noticed there is a web client for gazebo called gzweb http://gazebosim.org/gzweb.html and I'll try that next.

Generally, my question is, has anyone here thought of anything along these lines and maybe has some tips/tricks?

CSharpRon, 2020-07-31:
Hi @joerou! That's awesome to hear. 

We have not experimented with gzweb or thought much about using it in a classroom environment. Doesn't mean we can't now though!

Have you checked out the ROS Development Studio? It's a website where you can open entire ROS projects in the browser (including Gazebo). I know they have pricing tiers but it is an option: https://www.theconstructsim.com/teach-ros-effectively-v1-2/

Tduncan13, 2020-07-31:
The Jetson Nano dev kit is also pretty good. I have been experimenting with it myself and Ubuntu 18.04 has been pretty stable on it so far. You can pick up one of these boards for about $99.

tgsachse, 2020-08-01:
There's also AWS RoboMaker, I haven't messed with it much at all but I know it basically can wrap gazebo/Ros inside aws, so u could run the sim on their cloud servers. This would obviously cost money though. The Jetson Tyler mentioned is an interesting idea.

Do you want each student to be able to run the suite individually? Or do you want to do something like a class demo, where one person runs it on a projector/etc?

ROSs distributed nature means you can theoretically run the nodes whereever and have the headed simulation render on a different machine on the network which I think is along the lines of the VPS you mentioned. Surprised it was causing you trouble. Not sure if @CSharpRon did any work with cross network stuff. I think @camilozano actually may have run our graph on multiple boxes across a network but I'm not sure.

joerou, 2020-08-07:
Thanks everyone for the quick responses. Excuses for the delay!

I've taken a look at the suggestions above and had some discussions with the educators. It seems that some of the students don't have access to a computer at all (just an iPad), getting boards (either the newest raspberry pi with 8gb of RAM I think would work or the Jetson Nano dev kit) would be a cool idea, but perhaps slightly impractical at this point (everyone would need their own as you can't share anything during the pandemic and teachers are worried about an inability to give hardware support remotely).

So that, kind of narrows it down to getting it running with GZWeb, AWS RoboMaker or the ROS dev studio (I think this one is the most expensive out of the 3 though.). I'm working on seeing if I can get it going with GZWeb this week, if I have something worth pushing back I'll make the PR.

> ROSs distributed nature means you can theoretically run the nodes where ever and have the headed simulation render on a different machine on the network which I think is along the lines of the VPS [...]

The problem with this compared with the GZWeb implementation is that I'd still have to get the GZClient working on everyone's laptop (or in some cases iPad) rather than using a web browser, which looks like it would still require linux, so we'd save on processing power but not so much in getting everyone set up.

> Do you want each student to be able to run the suite individually? Or do you want to do something like a class demo, where one person runs it on a projector/etc?

It's looking like we may end up focusing on a specific piece of the robot (they have access to to fusion360 - should be able to export something comparable to blender?) and then we will run the simulation in the cloud and hopefully they can access it via the browser. We are working on a team based approach. If this doesn't work I'll check out the AWS RoboMaker. 

I'll keep posting here if you want to leave the discussion open, might be able to help the next guy :)