FlaSpaceInst/EZ-RASSOR #369: Wiki Refactor

CSharpRon, 2020-09-07:
Open issue to update the wiki (add bullet points as needed):

- Add documentation for Action Server usage
- Add documentation for keyboard teleop
- Refactor the wiki to have a topic-based hierarchy rather than just package information. Here's an example:
```md
Installation
-> ROS Installation
-> Workspace Setup
-> Package Installation
Simulation
-> Starting the simulation
-> Loading different Worlds
-> Spawning multiple robots
ROS
-> Topics exposed
-> Message Types
Control
-> Mobile App
-> Gamepad
-> Keyboard
-> Action script
```

tgsachse, 2020-09-12:
Keyboard teleop is done pending review. My suggestion for this overall is when we set up the new repos we should add the wiki page as the readme to each node, more or less. May also need to refactor them to fit the ROS wiki template system.

CSharpRon, 2021-01-25:
We should also add instructions or a tutorial for how to get this running on hardware.