FlaSpaceInst/EZ-RASSOR #374: Refactor configurable_simulation.launch spawn coordinate argument names

CSharpRon, 2020-09-15:
https://github.com/FlaSpaceInst/EZ-RASSOR/blob/65045ffda252fc17dfdfeb34581939c3c63a4fa4/packages/extras/ezrassor_launcher/launch/configurable_simulation.launch#L42

Maybe this is a non-issue but why does configurable simulation accept spawn_x_coords as input and it later gets passed to ezrassor_sim_description as spawn_x_coord (plural -> singular). It would be best if they were consistent.

tgsachse, 2020-09-17:
Yeah this is actually correct from my understanding:

In configurable simulation, you can spawn an arbitrary number of robots at once. You can configure each robot's spawn coords with `spawn_?_coords`. So if you wanted 3 robots, you might do this:
```
roslaunch ezrassor_launcher configurable_simulation.launch control_methods:=autonomy robot_count:=3 spawn_x_coords:="1 2 3" spawn_y_coords:="3 2 1"
```

The configurable_simulation launch file then recursively spawns all three robots. Each time, it calls the configurable_communication.launch file and also some other stuff (like the sim description). Each of these takes a single coordinate, because it is for a single robot.

So for robot 1, configurable_simulation.launch will get index [0] from `spawn_x_coords`, then pass it to configurable_communication.launch as `spawn_x_coord`. For robot 2, it'll get index [1], and so on.

Does this answer your question/concern?

tgsachse, 2020-09-17:
Here's the explanation in action: https://github.com/FlaSpaceInst/EZ-RASSOR/blob/mainline/packages/extras/ezrassor_launcher/launch/configurable_simulation.launch#L43

You can actually see all the args to configurable_communication.launch are "stepped down" from plural to singular, based on the index/robot count/robot id of the current recursive iteration of configurable_simulation.launch.

CSharpRon, 2020-09-17:
Woah, first of all thank you for the research. Second of all, that is some clean use of a launch file for this functionality.
I'll close this issue since there's a valid reason for the code.

closed 2020-09-17 by CSharpRon